iakdis/librecamera #1: Link to GitHub & Camera device buttons launch

PeterKW, 2022-10-03:
Looks really good! Thanks very much for this project, have been looking for a good open source camera application for a while and this looks really useful.

I'm not sure what the API is to set the app as a camera category, but it would be great to be able to set LibreCamera as the app to launch with the Android camera gesture.
[image]

Could there be a link to the GitHub repo inside the app? Perhaps in the about menu?

I'm more used to DotNet Maui for cross platform app development but I've done a small amount of flutter before.

Thanks again,
P

iakmds, 2022-10-03:
Hi Peter,

I am glad you like the app! Thank you a lot for your ideas, I have added both features (1. set the app's category as a camera, also resulting in the ability to set the app as the default camera app through the Android settings, 2. GitHub repository link inside the About menu) to the app!

You can download and try the attached APK file and let me know if everything is working. Once the first version of Libre Camera gets released to the F-Droid store, I will create a new release for everyone to download with all the updates. Thank you Peter.

[test-updated-librecamera.zip](https://github.com/iakmds/librecamera/files/9697309/test-updated-librecamera.zip)

iakmds, 2022-10-10:
Version 1.1.0 has been released, which implements the features – you can get it from the [releases](https://github.com/iakmds/librecamera/releases) page.

If you don't mind, I will close the issue, as the features are now implemented. Feel free to open a new issue if there is anything else!

closed 2022-10-10 by iakmds