iakdis/librecamera #2: Not working?

derpyderpaderp, 2022-10-04:
For some reason after taking the picture or video, I select it in the tiny window but it shows up as a non existing picture. I go into my built in gallery and it doesn't even show that it even exists. I have had it set to go to my DMCI folder on my internal storage and my SD card with the same issue. Even tried it outside of the DMCI folder for both storages. I've also tried giving it a personal storage with the same issue. Any reason why it would be doing this?

iakmds, 2022-10-04:
Hi there, thank you for the issue. That's interesting. 1. Could you check whether you have given the permission to use storage inside your Android settings? When I disable the permission, It seems that I get the same result as you. 

2. Also, what Android version are you using? 

3. Lastly, does the Save Path correctly show up in the app settings? In my case it is `/storage/emulated/0/DCIM`

microsuperman, 2022-10-04:
I am also having this issue. I am running android 10 on a Moto G7 Power. I have a permissions set to allow for camera, mic, and storage. With the path set to /storage/AFB8-7192/DCIM/Libre

iakmds, 2022-10-05:
> I am also having this issue. I am running android 10 on a Moto G7 Power. I have a permissions set to allow for camera, mic, and storage. With the path set to /storage/AFB8-7192/DCIM/Libre

Thank you for your info. I will take a look at this – I'll try a different saving mechanism on a phone with a higher Android version like yours. I am sorry for the inconveniences, but do stay tuned for the next releases!

iakmds, 2022-10-09:
Version 1.1.0 has been released, which should fix the issues – get it from the [releases](https://github.com/iakmds/librecamera/releases) page and please let me know if everything now works!

closed 2022-10-09 by derpyderpaderp