iakdis/librecamera #3: Feature Request: Manual Focus

JaceMakings, 2022-10-04:
Suggesting a manual focus function. Would help a lof with trying to take pictures of insects for iNaturalist.

iakmds, 2022-11-08:
(See the progress in this Flutter issue: https://github.com/flutter/flutter/issues/111476)

Devanand-PA, 2023-02-11:
That would be a great feature, because a lot of OEMs refuse to put in mannual focus especially in lower end phones. Even opencamera doesn't have this feature at the time of writing.
The only app on f-droid that I know has this feature is FreeDcam, which is highly unstable, has an outdated UI and old-fashioned sliders even for zooming, meaning no pinch-to-zoom.

Would be a great help to add this feature, especially for people who like to take photos of plants and insects!

Loughty, 2023-02-19:
> Even opencamera doesn't have this feature at the time of writing.

Actually, it has:

https://user-images.githubusercontent.com/52808915/219954338-d2ff0296-5a08-4efe-929c-e3805b962ff9.mp4

(here, a view of my very dirty keyboard :sweat_smile:)

Just make sure that you have "Camera2 API" selected on the "Camera API" option on the configurations, that give you access to more features. I hope this could be of help :grin: 

> That would be a great feature, because a lot of OEMs refuse to put in mannual focus especially in lower end phones

Yeah, the principal reason that I go to an alternative camera app was precisely the lac of manual focus. And was very disappointing that LibreCamera can't have that at the moment, because I like more this interface than the OpenCamera's one.

Devanand-PA, 2023-02-24:
> 

Thanks. That helps a lot. ^_^

oong819, 2024-05-26:
I got very frustrated when a tiny spider ballooning in the air but the OEM camera app only have auto-focus which focus into the wall instead of the tiny spider, waiting for Flutter to solve...