iakdis/librecamera #4: App requires network permission?

Glitchy-Tozier, 2022-10-05:
Fdroid detects this app requiring network -permission. Why is that?

If not needed, it would be great if unnecessary permissions could be removed.

[image]

Glitchy-Tozier, 2022-10-05:
If none of your packages require networking-permissions (which at first glance I don't think they do), this may be due to one of the the following:
1. You accidentally released the debug-build of your app
2. You might need to manually remove the requested internet-access from some xml-file in your android-directory

Glitchy-Tozier, 2022-10-05:
You're right, you seem to use the exact same files as I do in my app (that doesn't require internet access).
- https://github.com/iakmds/librecamera/search?q=internet
- https://github.com/Glitchy-Tozier/githo/search?q=internet

Also you seem to be using the same build commands I do:
- https://gitlab.com/fdroid/fdroiddata/-/merge_requests/11804/diffs
- https://gitlab.com/fdroid/fdroiddata/-/merge_requests/10627/diffs

No idea what is going on.

iakmds, 2022-10-05:
Interesting. Let's see if it will be there in the next release too. 

I wonder if it has anything to do with the packages I am using. Do you know whether packages in Flutter can effect permission use?

iakmds, 2022-10-05:
Well, that may at least be a hint, thank you.

Loughty, 2022-10-11:
Ey, @iakmds, I read the thread and for curiosity I made a quick search using the words "network" and "internet", and with the last one foud this in the source:

[image]

https://github.com/iakmds/librecamera/blob/2de3b697072c5e0ca44d76a49e9918a8e6d3cf96/android/app/src/profile/AndroidManifest.xml

https://github.com/iakmds/librecamera/blob/2de3b697072c5e0ca44d76a49e9918a8e6d3cf96/android/app/src/debug/AndroidManifest.xml

I hope this could be usefull :smile:

iakmds, 2022-10-11:
Hi,

thanks for the idea; Sadly the `AndroidManifest.xml` in these folders (profile and debug) are not merged with the final apk (it's a Flutter thing), therefore it cannot be the problem. What is interesting is that F-Droid marks the permission as `view network connections` which isn't even real "internet access", let's see and wait if it's still the same in the new version once it comes to F-Droid

Loughty, 2022-10-11:
:thinking:

Well, for now, I gonna leave this links from the F-Droid:
This is the source that F-Droid uses, maybe comparing both sources (from F-Droid and Github) could give to you some hint, I think there's programs that make the comparision more easy:
https://f-droid.org/repo/com.iakmds.librecamera_1_src.tar.gz
And this is the log of the compiling process:
https://f-droid.org/repo/com.iakmds.librecamera_1.log.gz

Also, I extracted the AndroidManifiest directly from the APK of the F-Droid version, and yeah, there's nothing about network or internet:
[AndroidManifest.txt](https://github.com/iakmds/librecamera/files/9754377/AndroidManifest.txt)
Maybe you could see something that I don't.

iakmds, 2022-10-11:
In the `AndroidManifest.xml` you extracted, it actually does say the following on line 13:

`<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE"/>`

So it is the thing F-Droid detected, we need to find out where it comes from (especially if it's still there in the next F-Droid version)

iakmds, 2022-10-27:
### Good news everyone 🎉
The next version won't require the permission for Network Access 🥳🎉

<details>
<summary><strong>For the ones interested</strong>, here is how I fixed it:</summary>
The cause for the Network Access permission request was the package <code>video_player</code> (to my surprise), which I used for displaying the video preview/thumbnail. I discovered this by creating a new blank project and adding all the packages from Libre Camera. I then removed the packages in different combinations, built the apk and inspected it's AndroidManifest for the permission. And after a couple of tries, the <code>video_player</code> package was left.

Now, the app will be using the <code>video_thumbnail</code> package instead – no need for any permissions 🎉
</details>

Glitchy-Tozier, 2022-10-27:
Great to hear, good job! 🎉

iakmds, 2022-11-02:
Version 1.3.1 is now on F-Droid, and as you can see, the permission is gone!
[image]

Can we close this issue? :tada:

Glitchy-Tozier, 2022-11-02:
Sure :)

closed 2022-11-02 by Glitchy-Tozier