iakdis/librecamera #7: Perhaps a stupid question how to quit the app ?

Npa972, 2022-10-06:
After opening the app i cannot close it (no  menu to go back or close).

iakmds, 2022-10-06:
Hi there,

I suppose you are talking about the missing navigation bar at the bottom of the screen. On Android, you typically press the **navigation buttons at the bottom** of the screen. **However**, when the app is full screen like in this case, they are **hidden** so that the camera app gets as much screen estate as possible. You can **drag up from the bottom** to make the buttons visible again and simply use the back or home button.

Does this help you? If not, don't hesitate to ask further questions.

Npa972, 2022-10-06:
Thanks, no problem it's working as you said !

iakmds, 2022-10-06:
Awesome!

This will be the first closed issue on this repo, congrats :)

closed 2022-10-06 by iakmds