iakdis/librecamera #12: Fixed my previous issue

derpyderpaderp, 2022-10-09:
Okay 1.1.0 fixed my previous issue of not being able to open the file/s, thank you for that. But now I can't open the picture via the little capture preview. I have to leave the app and go to my gallery app. Love the feel of this app btw, feels so smooth and the flow of it is amazing.

iakmds, 2022-10-10:
Hey,

I am glad the issue got solved and that you like the app. With regards to the little capture preview/thumbnail, I am aware of the issue, as noted in the release notes under "Known issues" and will fix it soon. I wanted to get the new version out, so that saving to the gallery (arguably the most important thing ...) works properly. Out of interest, what Android version are you using?

iakmds, 2022-10-15:
Hi, version 1.2.0 has been released, which should fix the issues – you can get it from the [releases](https://github.com/iakmds/librecamera/releases) page and please let me know if everything now works!

derpyderpaderp, 2022-10-15:
Works great now. Thank you for this. Been looking for a camera app like this forever.

closed 2022-10-15 by derpyderpaderp