iakdis/librecamera #15: The "camera from locked screen" funtion is broken only when I have installed the app

Loughty, 2022-10-11:
Here an example:
[broken.webm](https://user-images.githubusercontent.com/52808915/195056403-e0e7a2fa-fdbb-46d2-aaed-e107e0a245f3.webm)
And this is the expected behavior:
[normal.webm](https://user-images.githubusercontent.com/52808915/195056542-47ea4d4e-49b1-45ec-8c4e-7dea90477e39.webm)

I am not a developer, so I don't know really what is the cause, but I think have something to do with this:
https://github.com/iakmds/librecamera/issues/1
Because when I use the F-Droid version (which still at 1.0) this problem does't happen. If that the case, maybe a swith to deactivate the funtion of being "reconice has a camera" could work for now.

My phone uses Android 12 Go, if that helps :smile:

iakmds, 2022-10-11:
Thanks for the issue. 

How weird, do you have selected Libre Camera as your **Standard Camera App** or is your default camera selected?

Loughty, 2022-10-11:
That whas the first thing that I checked, and extrangely, there is not "default camera app" option or something like that:

[image]

Just to make sure, I check with another phone which have Android 10 Go, and the same problem:

[image]

I don't know if that something about the "Go" edition of Android :thinking:, but is a little bit anoying :disappointed:

Loughty, 2022-10-11:
(Click on the wrong button, sorry)

iakmds, 2022-10-11:
No problem.

Can you try to remove Libre Camera and install another camera application (like Simple Camera on F-Droid) to see if the problem has to do with Libre Camera or with the Android Go version and a different camera in general?

Loughty, 2022-10-13:
Yep, is the same when I install another camera app, but, I read this on the Simple Camera description:

[image]

Basically, says that the camera button doesn't work on some devices unless you deactivate the default camera app, and I think it's the same with another camera shortcuts, like the “press the power button twice” or the “open camera from the lock screen slicing the right corner” (my case). Unfortunately, I couldn't probe this, because I can't deactivate my default camera app…

[image]

I think anyone who has the same restrictions (can't change the default camera app and or can't deactivate the default camera app) could have the same problem.

iakmds, 2022-10-13:
Hi, thanks a lot for the information. This basically shows that it's not something specific about Libre Camera but rather about a different Camera app in general that has problems with the default one. 

If you say however that it did work with the first version, I should add a new field in the settings that stops the app from being seen as a camera like you said ("_Because when I use the F-Droid version (which still at 1.0) this problem does't happen. If that the case, maybe a swith to deactivate the funtion of being "reconice has a camera" could work for now_")

jld3103, 2022-10-26:
I'm not sure if my problem is the same one as this one, but here it is: I have a shortcut set to open the camera when double clicking the power button. It should work when the screen is locked too, but not with Libre Camera. Open Camera has a setting for it and it works there. I'm not sure if this functionality should already be present in Libre Camera and this is a bug or if I'm requesting a new feature right now. Please let me know if I should open a new issue.

iakmds, 2022-10-26:
Hi,

thanks for commenting. This could actually be the same problem, though I'm not sure about this. Thanks for letting me know that it works on Open Camera. I'll have to research how to implement such feature. No need for a new issue for now.

To @Loughty: You said the problem also occurred with Simple Camera, it would be great if you could try Open Camera as well (In Open Camera: Settings → More camera settings → Along the lines of "Show camera despite lock"/the 4th last option)

Loughty, 2022-10-26:
> To @Loughty: You said the problem also occurred with Simple Camera, it would be great if you could try Open Camera as well (In Open Camera: Settings → More camera settings → Along the lines of "Show camera despite lock"/the 4th last option)
Yep, same problem, the shortcut doesn't work, with or without that option activated, but the function in the description work has expected, the app keeps open even when you lock the phone, like the Google's Calculator.

https://user-images.githubusercontent.com/52808915/198140215-04cf7f42-42be-4672-a38c-f47d925af1ed.mp4

SamTada, 2022-11-03:
Hi, related issue here?
I'm on MIUI 13.0.15 (Xiaomi's equivalent to Android 12) and I had the system camera 'uninstalled' via ADB. Been using Open Camera since and recently installed Libre Camera, hoping to switch.
This is what is shown in the Default Apps in MIUI settings and in Android settings. In essence, nothing is recognized.

[image]
[image]
[image]

Also used Better Open With and Default App Manager to try and redirect to any camera, but only the latter listed compatible camera apps though solely for video.

[image]
[image]
So, would it be possible to create a dedicated Libre Camera quick button for the Control Panel – similar to what Open Camera has? For the time being, I obviated this with a non-descript tile from Quick Tiles.

[image]

Could you look into this please? Thanks in advance.

iakmds, 2022-11-04:
@SamTada:
Thanks for your information, I will look into it.
As for the Quick Tiles: I've created a [new issue](https://github.com/iakmds/librecamera/issues/30) for that and will add the feature.

SamTada, 2022-11-04:
@iakmds 
Thank you for replying so promptly.
Looking forward to the tile feature  ツ

iakmds, 2022-11-05:
@SamTada
That was easier than I expected: The Quick Tile feature is now included in the apk below.
The important thing now: Could you check if Libre Camera gets detected as a Camera app, or is that still not working? Thank you.

[app-release.zip](https://github.com/iakmds/librecamera/files/9943675/app-release.zip)

provokateurin, 2022-11-05:
i think it was correctly detected as camera app for me, because when i still had open camera installed and triggered the camera shortcut i was asked which app it should open. i think there is just no way to set the default camera app inside the android settings, because i also searched for it and couldn't find it.

iakmds, 2022-11-05:
@provokateurin 
Interesting, on all the devices I tested the app was detected as a camera app too. 

In regards to the other problem, the Android 7.0 and Android 11 devices that I tried could set a default camera app, but not an Android 12 device. Seems like there isn't a setting for that on every Android phone. Thank you.

provokateurin, 2022-11-05:
Interesting, I also run Android 12

SamTada, 2022-11-13:
@iakmds 
Sorry for the late reply.
Installed the uptade and yes, the Quick Tile works perfectly!
[image]
Libre is now my default camera – even if the OS doesn't acknowledge that still...
╮( ◕‿◕ )╭

iakmds, 2022-11-13:
Great to hear, thank you! A new update will be available soon which includes the quick tile feature, a redesign of the camera page and more. Stay tuned :tada:

WildRikku, 2025-06-01:
I think I have an issue that is very similar or the same. I use the "tap power button twice to open camera" functionality. It works with LibreCamera when the phone is unlocked (the first time it asked which app I want to use, I selected LibreCamera and now it's fine). However, it does not work when the phone is locked. The phone vibrates and the screen turns on and the notifications are hidden, but the camera does not appear. When I then manually unlock the phone, LibreCamera is open and shows a loading indicator instead of the camera image and the "an app is accessing your camera" indicator is not on.

Deactivating the default camera does not help.

I have LibreCamera 1.9.1 installed via F-Droid. I have Android 14 on a Fairphone 5.