iakdis/librecamera #20: leave bottom back / menu buttons up or have an option for it

G2G2G2G, 2022-10-16:
The bottom of some android phones have software (on screen) back / menu / home buttons or whatever they are. This app overrides those and you can't get out without pulling up the menu and sometimes it won't pull up.
I have this in my main android settings to "always show" but it doesn't on libre camera. Please at least have an option to always show in your settings if not hard coded

Thanks!

iakmds, 2022-10-16:
Thanks for the idea, I will add a new field inside the settings, probably under the "More" section. I already received a comment about not seeing the navigation bar, so this should be a good addition. Thank you

iakmds, 2022-10-22:
Oops, I didn't know the issue would get closed if I mention it in a PR. Before I'll close it, I'd like to get confirmation from you once the new version comes out

G2G2G2G, 2022-10-22:
@iakmds lmao yea gitea has this just discovered it a few days ago coincidentally, with the keyword "close / closes #IssueNumber" 
I guess github has it for "fix #IssueNumber"  (probably close/closes too)

iakmds, 2022-10-30:
Hey @G2G2G2G, the option is now present in version 1.3.0, may I close this issue? Thank you

G2G2G2G, 2022-10-31:
awesome. works well, thanks

commit fadfe821 (2022-10-22): add: Option to show navigation bar to fix #20

Fixes #20
closed 2022-10-31 by G2G2G2G