iakdis/librecamera #21: camera video mode bit rate and resolution

G2G2G2G, 2022-10-16:
Currently seems to only have resolution and jpeg encoding compression settings.

Phone camera apps usually allow you to select which resolution, the resolution is hardware encoded by the camera module so only a few specific resolutions are usually supported.. but please list them and allow to select. 
Generally you'll have a few options like 1:1 and "full video" which does some weird size.. but majority of the settings will be 16:9 which are hardware supported and things like:
- 1280x720
- 1920x1080
- 1920x1080 60fps
- 2560x1440 (rarer than the others)
- 3840x2160

Older samsung phone camera apps allowed selecting the bit rate as well, which is very handy. Certain apps have file size limits so setting your phone to do 500kbps massively decreases the video size... generally the bitrate in phones is far overkill since their lenses are literal trash, so you end up with massive file sizes. But I'd like several minute videos being kept only several megabytes to make it easier to share.

Thanks!

iakmds, 2022-10-16:
Thanks for the issue. Basically, I am quite limited by the Flutter camera. There are 2 open issues on the Flutter camera GitHub repo: [Allow custom resolution and aspect ratio](https://github.com/flutter/flutter/issues/45665) and [Set custom frame rate and bit rate in Camera plugin](https://github.com/flutter/flutter/issues/54339), so I can't do much; Sorry about that.

I will leave the issue open however, maybe I'll be able to implement the features somehow in the future.

G2G2G2G, 2022-10-16:
cool thanks. That one comment "this is a major oversight" yea.. it definitely is sad to see google ignore that from the dates.

I'll just setup a bot to spam google employees emails with those thread URLs, thanks.
Guess I'll have to keep using ffmpeg. Thanks for the response & keeping it open.

PanderMusubi, 2022-11-13:
Even if changing fps is not yet possible, please list the fps the camera is currently operating at.

Pantyhose-X, 2023-09-17:
LOW resolution should be 720X480 resolution, 720P below the picture blur, on the eyes big damage