iakdis/librecamera #22: Can't choose storage path

hamid255, 2022-10-18:
My phone doesn't let me choose a storage path at the start of the app. When pressing the choose path button there's just a page with no directory on it:

[image]
[image]

my android's version: 5.1
app's version: 1.1.0, also existed on earlier version 1.0.0
downloaded from f-droid

iakmds, 2022-10-19:
Hi, thank you for the issue.

That's weird, what happens though, if you press the hamburger menu at the top left? It says "Recent" in the top left, maybe there aren't recent folders and you have to select the phones storage path and go from there. Let me know if that works

hamid255, 2022-10-19:
the hamburger menu is opened at first, and only option shown is "recent" which is empty.
[image]

Loughty, 2022-10-21:
Sounds like you don't have storage permissions, do you check de "allow" option when Android ask it?

[image]

If not, or simply don't show anything, then go to "Apps" on Settings, select LibreCamera and manually give him storage permissions!

[image]

Try and see if that solve it :D

hamid255, 2022-10-22:
In Android 5.1 apps don't ask for permission, they either have it or not, and Libre Camera in my phone has it:

[image]

I think on/off switches for permissions were added in settings in Android 6. I don't have that.

greyspottedfrog, 2022-10-22:
Hi, i found a bit the same problem.
App version 1.1.0 and now 1.2.0 
On Android 11:
I can choose the path internal/DCIM which works and i can choose internal/newphotos as well but here it does not work. The app can not save the pic. Btw, also the folder DCIM had to exist, the app didn't create it.
Hope this helps, froggy.

greyspottedfrog, 2022-10-29:
Still the same in 1.3.0
and also the shutter sound does not work :-(

iakmds, 2022-10-29:
Hey,

I'm sorry for that. I don't know where the storage issue may be coming from, I have tested the app on Android 10, 11 and 12 and it was working there... Could you explain your exact issue again, I don't think I fully understand the last part `"I can choose the path internal/DCIM which works and i can choose internal/newphotos as well but here it does not work"`, what do you mean with "here"?

Regarding the shutter sound, that's also unfortunate, if someone else has this issue too I can create a new issue for that.

Thank you.

greyspottedfrog, 2022-10-31:
Hi,
With here i meant:
When i used the camera with storagepath internal/newphotos the app does not write the image.
Further there's no notable difference at all between the storagepath internal/DCIM and the path internsl/newphotos. The only difference is that there is no save.
Hope this explains, froggy

gadzhi-m, 2022-11-01:
Hi. I have a similar problem. On the second step I can't specify the path, I don't get any more windows.
Xiaomi rn5, firmware RR, android 10, Libre Camera 1.3.0

iakmds, 2022-11-03:
To @hamid255 and @gadzhi-m:
Would you like to try out an apk that includes a button sets the path `storage/emulated/0/DCIM` as the save path? If that works, it would be an issue with the file picker and I could include such button/text field to type in the path in the next version

hamid255, 2022-11-04:
Sure, I would.

iakmds, 2022-11-05:
Thank you. Let me know if it works, here it is:
[app-release.zip](https://github.com/iakmds/librecamera/files/9943667/app-release.zip)

hamid255, 2022-11-05:
It does work. Thanks!

iakmds, 2022-11-06:
Great. Thanks for your reply and your patience. I will probably add a text field for the next version (with the DCIM path as a placeholder) so others with this problem can still set a custom path

iakmds, 2022-11-14:
Hi again. A new release is out: v1.4.0, which includes a text field to set the path. I will close this issue, in case there is anything else, please let me know :tada:

closed 2022-11-14 by iakmds