iakdis/librecamera #23: Feature Request : Launch Selfie Camera By Default

DarkArrow9, 2022-10-20:
Hello,
Can you add an option in settings called 
Launch Selfie Camera at start/Default Camera Mode - Selfie ?

It will be great help for some people who don't want to use / can't use Rear Camera.

Thanks in advance.

iakmds, 2022-10-20:
Hey DarkArrow,

this option already exists inside the settings, does it work for you?

[image]

DarkArrow9, 2022-10-21:
> [screenshot](https://user-images.githubusercontent.com/111225446/197025379-4957f5d6-c8ba-4ef7-bf64-b81c455d306c.png)

This function didn't worked for me. Thanks for noticing. When I will get a new phone LibreCam will be my Default Cam app. I can assure you that.
-------- Original Message --------

DarkArrow9, 2022-10-21:
I forced stopped it, cleared the cache of it. Started the app and the front cam works!
You are truly a Fantastic, your work too!

It saved me, you can close this issue. ❤️

Thanks a lot for creating such a masterpiece! Love From 🇧🇩.
-------- Original Message --------

iakmds, 2022-10-21:
Awesome, thank you a lot my friend! 😄

closed 2022-10-21 by iakmds