iakdis/librecamera #27: Translation with weblate.org

nautilusx, 2022-10-30:
Thanks for your great app.

Can you please use weblate.org for the translation. It makes translation much easier for non-developers. It's free and open and for free apps, like yours, it's also free and gratis.

https://weblate.org/hosting/

iakmds, 2022-10-30:
Hi,

thank you, I totally agree with you! It is also listed on the roadmap: `More supported languages (See Contributing below) and Weblate integration"`
However, the project has to be active for at least 3 months – this one is a little over a month old (First commit on Sep 17). Once it reaches those requirements, I will integrate it!

iakmds, 2022-12-23:
Hello again,

I am happy announce that [Weblate](https://hosted.weblate.org/engage/librecamera) integration has been completed! 
For more information, check out [#translate](https://github.com/iakmds/librecamera#translate) in the README of Libre Camera. 

Thanks again and have a great day.

closed 2022-12-23 by iakmds