iakdis/librecamera #35: Feature request : camera api

woaloo, 2022-11-07:
Hi,
Could it be possible to add an option to sélect wich caméra API is using ?
Caméra api2 give more possibilities when shooting.
Wide angle is an exemple.

nfk, 2023-01-03:
Hi, by default which camera api is used with librecamera [cameraX](https://developer.android.com/training/camerax) ?

iakmds, 2023-01-03:
Hey,

I don't think the Flutter camera currently uses CameraX, though I am not sure. There is a [camera_android_camerax](https://github.com/flutter/plugins/tree/main/packages/camera/camera_android_camerax) GitHub repository in the Flutter camera plugin, which had its [first development commit](https://github.com/flutter/plugins/commit/0134dff6cd5744461e0c5f08f32300c9e648a073) on August 15, 2022, so it may be the future plan to migrate to CameraX

nfk, 2023-01-03:
Thanks @iakmds, by default it seems that flutter camera plugin uses the [camera2](https://github.com/flutter/plugins/blob/e85f8ac1502db556e03953794ad0aa9149ddb02a/packages/camera/camera_android/android/src/main/java/io/flutter/plugins/camera/Camera.java#L13) API. It should be great to see what cameraX bring with librecamera. Let me know if you have a test build one  day. I can spend sometime to test it.

iakmds, 2023-01-04:
Absolutely, thank you :+1: