iakdis/librecamera #36: Flash in Video mode

iiitomo, 2022-11-15:
First of all, thanks for this new app, I really like the default interface, it shows only the icons I need.

Currently, the Video mode has a Flash icon that behaves the same as for the Camera mode:
1) No flash
2) Forced flash
3) Automatic flash
4) Light
It makes sense for the Camera mode, but for the Video mode only the 1 (no flash) and 4 (light) are useful.
I suggest to remove the 2 and 3 (forced flash and automatic flash) for Video mode.

iakmds, 2022-11-15:
Hi, that makes a lot of sense. Thank you for this – will be added.

iakmds, 2022-11-29:
Hello again, the Flash Mode selection in video mode is now limited to off and torch in the new version 1.5.0 :tada:

I will close this issue now, in case there is anything else, do let me know.