iakdis/librecamera #40: Distortion with devices with cutout

RestlessDevil, 2022-11-20:
[image]

My phone is Xiaomi with display cutout and I have switched navigation bar to always present. The bottom portion is great, but the cutout is causing a problem (screenshot).
I would like to suggest limiting the app to the area under the cutout, so that both can coexist, or hiding the upper portion of the navigation bar alltogether.
Great app anyway, I have switched to it from OpenCamera.

iakmds, 2022-11-20:
Hi, thank you a lot for this issue. 

On my phone it got close to covering the top controls, yet I didn't notice it. This will be fixed in the next release. I'm glad you like the app!

iakmds, 2022-11-29:
Hello again, the app is now limited to the safe area of the screen in the new version 1.5.0 :tada:

I will close this issue now, in case there is anything else, do let me know.

closed 2022-11-29 by iakmds