iakdis/librecamera #42: LibreCamera can't save pictures

iiitomo, 2022-11-20:
LibreCamera 1.4.0 fresh install asks permission for using camera and mic, but not permission for writing on memory.
Until I manually gave write permission to the app, it wasn't able to save pictures.

iakmds, 2022-11-21:
Hi, thanks for the issue. Could you tell me which Android version you are using? Thanks

iiitomo, 2022-11-22:
Lineage 10
Fyi, if i uninstall/reinstall LibreCamera, the bug replicates.

iakmds, 2022-11-22:
Hm, interesting. With Lineage 10 you mean Lineage OS that has Android 10? Would you also tell me if you do get the storage permission request on other apps, for example Open Camera/Simple Camera? If not, maybe it has to do with your device, if yes, the request is probably simply not working on this app

iiitomo, 2022-11-23:
Yes, lineage17.1 that has android 10.
After reinstall, SimpleCamera asks the Storage (and camera/mic) permission, and works fine (pictures are saved).
After reinstall, OpenCamera doesn't ask Storage permission (only access to camera) but works fine (pictures are saved).

iiitomo, 2025-02-27:
Bug still valid with current Fdroid version (1.9.1).