iakdis/librecamera #43: Screen doesn't stay on

Cwpute, 2022-11-22:
Hello,
I'm having trouble using LibreCamera, often times the screen fades to black because i don't interact with the screen often enough.
Is it a default behavior or is it something vrong with my phone that i could fix ?

If it's a default behavior, i'd like to suggest an option to keep the screen on when using the app, or double its lenght or something for this app only.

iakmds, 2022-11-22:
Hi, thank you very much. Nothing is wrong with your device, it's Libre Camera that needs to keep the screen awake. Because I had my display awake time set to 10 minutes I have never noticed that. Setting it to 15 seconds and you're right, that shouldn't be happening.

It will be fixed in the next release, thank you!

iakmds, 2022-11-29:
Hello again, the screen is now always on in the new version 1.5.0 :tada:

I will close this issue now, in case there is anything else, do let me know.

closed 2022-11-29 by iakmds