iakdis/librecamera #49: Fixed Focus doesn't work

Cwpute, 2022-12-31:
The camera keeps trying to focus automatically in Video mode, even when Fixed Focus is selected.
I'll try other apps to see if this is a problem with my phone or with this app.

I'm using a Nokia TA-1157 with Android 11.

Cwpute, 2022-12-31:
Tried the feature on FreeDCam and fixed focus does work there.

iakmds, 2022-12-31:
Hi there, thank you for your issue. 

I tested it and can confirm it. Though I'm afraid this is another one of those things I have little control over as it is handled by a plugin – let's see what can be done.

shodanx2, 2023-02-12:
Hello,
I think I have elucidated this problem.
The auto/locked setting in the app, the labels are reversed

Observe this behaviour, when using tap to focus, with focus set to auto, it will not respond, but it will work when it's locked !

This is a on a Moto Z4

https://user-images.githubusercontent.com/10621885/218286629-dc44ef0f-9e74-472a-bc0e-d151847687d7.mp4