iakdis/librecamera #51: Can't save to SD card

maverick74, 2023-01-08:
LibreCâmera asks for permission to have access to storage in the wizard and, even granted, if I configure it to save pictures to the SDcard it doesn't!

It takes the picture but the white frame surrounding the image never goes away and the image is never saved.

If it's configured to save on the phone memory it works well!

Please, fix this :)

maverick74, 2023-01-18:
@iakmds I would like to ask 2 off topic questions. Should I open an issue or can I ask here?

iakmds, 2023-01-18:
You can create a new issue and I'll assign the "Help wanted" label

maverick74, 2023-01-19:
@iakmds I would like to ask 2 off topic questions. Should I open an issue or can I ask here?

> You can create a new issue and I'll assign the "Help wanted" label
> 
> *Also, regarding the SD card issue, may I know which Android version you are running?

I'm on Android 8.1.
if you need to know something else I'll be around (No, I don't know the lottery numbers! XP  )

ksnnkr, 2023-02-25:
I am facing the same problem.
Android version is 9.
Even though I have given permissions regarding the SD card, it gives me the message as shown in the attached image and I am unable to save it.
It used to be the case in my environment.

>It takes the picture but the white frame surrounding the image never goes away and the image is never saved.

But today, after updating to version 1.6.0, it now gives the error message as shown in the attached image.

[image]

iakmds, 2023-02-25:
@ksnnkr thanks for your info. Is the path `/storage/6CD1-3126/DCIM/` a path on the SD card, or is it on the phone in this case?

You're right, the new version includes some error messages when things go wrong. What this error says is that it cannot copy the image file from the cache to the final destination; Probably because of an SD card related permission issue when saving that needs to be fixed

ksnnkr, 2023-02-26:
Sorry for the lack of words.
/storage/6CD1-3126/DCIM/ is the path on the SD card.

In addition, today I tried to disable the permissions related to storage, and the message shown in the attached image appears **at startup**. The screen is dark because the cover is still on.
[image]

mega-stoffel, 2023-04-26:
same issue here. 
Libre Camera can't write any files, neither to SD card nor to the internal storage.

Running Lineage19.1, and other apps (like Open Camera) are working fine.

Screenshots from the (pretty similar) error messages:

[image]

[image]

rgabbo, 2023-05-17:
I can confirm this issue also in Android 10

sillyrat, 2023-06-21:
Same issue here with GrapheneOS (Android 13): "Operation not permitted" errors when trying to access internal storage.

jdenm8, 2023-08-17:
Confirming on Graphene 230805 on a Pixel 6a. The Give Storage Permission button in the Onboard Wizard does nothing and manually pointing the app at a directory throws the previously posted 'Operation Not Permitted' error.

EllesarDragon, 2023-11-04:
I have a samsung galaxy J5 2017(yes I know, but was practically forced to get a samsung phone, since only unaltered samsung phones where allowed).
android 9
app version 1.9.1

I could give t the permissions from the settings, 
saving images to internal storage does work, but when saving to external storage it gives a 
PathAccesException: Cannot copy file to '/storage/sdIDhere/libcam/imagename.jpg', path = '/data/user/0/com.iakmds.librecamera/cache/otherimagename.jpg' (OS Error: Permission denied, errno =13)

perhaps it is related to t[](his: https://stackoverflow.com/questions/60558577/how-to-solve-os-error-permission-denied-errno-13-in-flutter [)](https://stackoverflow.com/questions/60558577/how-to-solve-os-error-permission-denied-errno-13-in-flutter)
which is a problem which happens due to a conflict between before and including android sdk28 and android sdk29 and later.
android sdk 29 is aimed for android 10 and later and uses a difference in how it handles and requests external storage
they say that adding:
"
<application
      android:requestLegacyExternalStorage="true"

"
to the AndroidManifest file would fix it
however would need to see if that doesn't affect newer devices, and if it does it would only be worth it if it only enables that setting on older devices, or if it has to be manually enabled.

EllesarDragon, 2023-11-04:
I see you already used that setting in the main AndroidManifest file.

in my case the first path it shows in the error is a correct one, the /data/user path I do not know, I think that is cache or ram where the image is stored before writing to normal storage or such.

FreeDCam does work on my device and is also opensource.
https://github.com/KillerInk/FreeDcam/blob/60682b2a086988f291725f8c9866a41350abda91/app/src/main/AndroidManifest.xml#L4
so perhaps you can find the solution for the problem there.
their AndroidManifest file does not seem to use that line which is supposed to fix this issue, so perhaps that one line might be the cause of the problem?

or perhaps a older android apk version is used there?(I don't know which versions either projects uses)

InternetOutlaw, 2024-04-21:
Add me to users w/ same issue. LibreCamera works fine on a Samsung Galaxy A04e, SM-A042MS/DS, but not a Samsung Galaxy A12, SM-A125 U.  PathNotFoundException: Directory listed failed path=storages/emulated/0/DCIM (OS Error: No such file or direcory erno=2. Creating a path on the SD card or in the phone results in failure. Have tried multiple paths. Just installed Open Camera but its not Libre Camera which is perferred.

NWJ9PB, 2024-10-18:
Bumping this issue, 
Issue still present at current FDroid release

iiitomo, 2025-02-27:
Variant even worst for me: in addition to failing on SD card, it fails writing on internal memory too. (See bug #109)