iakdis/librecamera #53: Privacy by default: make microphone permission optional

staticdev, 2023-01-18:
This is a wonderful app and I would like to make it a bit better. If we follow the principle of least privilege, users that only take photos with camera app should not be mandated to allow microphone permissions to take their first photo. I would like it to be optional following the standard of other camera apps and requiring the permission only when the user asks for video.

iakmds, 2023-01-18:
Hi there,
thank you a lot. This makes a lot of sense. Will be implemented in the next version.

iakmds, 2023-02-24:
Hello again @staticdev and the others,
it took a little while but here it is: v1.6.0 is now available on F-Droid! (and GitHub)

Let me know if I can close this issue :smile:

staticdev, 2023-02-24:
> Hello again @staticdev and the others, it took a little while but here it is: v1.6.0 is now available on F-Droid! (and GitHub)
> 
> Let me know if I can close this issue smile

@iakmds I tried now to install from F-Droid, but latest I see is 1.5.1.

closed 2023-02-24 by staticdev