iakdis/librecamera #55: Feature Request: Slow motion

d92t8b1i, 2023-01-20:
In order to utilize the slow motion capabilities of my Pixel phone's camera I have to use the Google camera app. It would be amazing if I didn't have to leave the libre camera app. Is it possible to add this feature?

iakmds, 2023-01-20:
Hi,
thank you for the feature request. It looks like it is possible [(https://github.com/android/camera-samples/tree/main/Camera2SlowMotion)](https://github.com/android/camera-samples/tree/main/Camera2SlowMotion), but don't expect this feature to be implemented in the _very_ near future, as there are also other bugs/features I am working on and this isn't as big of a priority in comparison to other things (but maybe there will betime in the future)

*Additional note, you _technically_ wouldn't have to _leave_ Libre Camera altogether just for Slow Motion, you could also use Libre Camera as your main camera and only use the Google camera for Slow Motion... :wink:

d92t8b1i, 2023-01-20:
Thank you for your reply and honesty. I look forward to the feature, even if it takes time. Thank you.

Regarding leaving libre camera to use slow motion in Google camera, I simply meant it would be nice not to have to use both camera apps. By "leave" I meant closing the app to open Google camera.

Thanks again!