iakdis/librecamera #57: An resulation option for video and photos independently

CoolerWuffi, 2023-02-03:
Hello, 
Idk if any requestet something like this befoe me. 

It woud be awsome if the user would be able to choise the resulation of thier photos and videos independently. 
I personly want to have the max resulation what my phone can for my photos and for videos only full hd cause 4k is extremly memory expensive

iakmds, 2023-02-03:
Hi there,
Thank you for the idea. Yes, I think I understand what you mean. I'll add this in the upcoming versions, maybe already in the next version :+1:

Burys14, 2023-07-17:
Hello,
please add this function. It is really not very good to take the pictures in only 2 megapixels resolution. The app seems very promising for me, but I really need to set the meaningful resolution for the pictures. Phone Motorola Moto Z2 Play with LineageOS 18.1.
Thank you.