iakdis/librecamera #59: [Question] Would a streaming feature (eg to OBS) be possible?

SmutnyStefan, 2023-02-17:
Every single app that does this is closed source and paid or is severely limited

iakmds, 2023-02-17:
Hey there,
thanks for the question. Interesting feature, but I think it is out of the scope of Libre Camera... I try to put the focus on making it a nice camera app.
Too bad that there isn't any viable option out there at the moment, let me know if you find one though!

(Will close this issue, but feel free to comment more)

closed 2023-02-17 by iakmds

HIHIQY1, 2023-02-27:
Hey, I had the same problem! What you could do as a workaround is use [scrcpy](https://github.com/Genymobile/scrcpy) if you're on Android, then add an OBS window capture which captures the scrcpy window :)

Ernest1338, 2023-05-18:
@SmutnyStefan Have you tried using https://vdo.ninja/?