iakdis/librecamera #60: [Feature request] Switch to Material You

piu130, 2023-02-21:
**Describe the solution you'd like**
Migrate the app from Material to Material You. Flutter seems to have build in support for Material You, but it's opt in.
If you want to wait until it's default, at least an adaptive icon would be great, so the homescreen would be more consistent :)

iakmds, 2023-02-21:
Thank you a lot for this issue! I was aware of Material You, somehow I didn't consider it yet.
Material You will be enabled by default in the next version, it can be disabled via settings. 

Regarding the adaptive icon, it should also work in the next version, though we'll need to test that

piu130, 2023-02-24:
Wow. That was fast :) Thank you!
The icon does not currently work on my device, but this might be a problem with with the library `flutter_launcher_icons`.
See [this issue](https://github.com/fluttercommunity/flutter_launcher_icons/issues/458).

closed 2023-02-24 by piu130

iakmds, 2023-02-24:
Right, I also saw another issue regarding adaptive icons in that package. Anyway, hope you enjoy the update!