iakdis/librecamera #63: [Bug] Pictures do not appear in Gallery

mvglasow, 2023-03-03:
**To reproduce**

1. Set image location to `/sdcard/Pictures` and ensure permissions are granted.
2. Take a picture.
3. Tap the picture to view it in Gallery.
4. Return to the Android home screen.
5. Launch Gallery individually, and try to find the picture.
6. Open a file manager and look for the picture in its folder.

**Expected behavior**

The picture appears in Gallery, just like pictures taken with other camera apps.

**Actual behavior**

While I can view the picture in Gallery by tapping it right after taking it, it no longer appears when I launch Gallery and search for it – neither in Timeline nor in Albums. However, the file does get created, which can be verified with a file manager. I can also tap the file in the file manager to view it with an app of my choice.

**Other Information:**
 - App version 1.6.0, from F-Droid
 - Device: Fairphone 4
 - OS: Android 13 (TQ1A.230105.001.A2), LineageOS 20-20230129-NIGHTLY-FP4
 - Gallery app: `com.android.gallery3d`, version 1.1.40030 (included with LineageOS 13)

Behavior may be different with other gallery apps, since this is likely  due to how the gallery app detects the presence of new image files. However, this is a fairly standard gallery app, which so far has worked with any other camera app I tried – which indicates troubleshooting should begin on the camera end. After rebooting the phone, the new pictures still do not show up.

My first suspicion would be that LibreCamera fails to trigger a media scan after taking a picture, or the scan does not complete for whatever reason. (As far as I remember, there are two ways to trigger a media scan: one through a Java API call, another as an intent. I had an app where I needed to scan whenever the app crashed – the Java API method did not work, presumably since the app terminated before the scan finished, but the intent method worked.)

iakmds, 2023-03-03:
Thank you for such a detailed description.

I would love to know, does this issue apply to a non-SD card save path? There is another issue regarding SD card support, which may have to do with this one. If not, it probably has to do with the MediaScan...

mvglasow, 2023-03-04:
As for SD card, `/sdcard` somewhat misleadingly refers to the internal shared storage of the device, which nowadays is usually built into the device rather than a removable SD card. `/sdcard/Pictures` (on earlier Android versions, `/sdcard/DCIM`) is the default path for camera pictures.

So, in my scenario, no actual SD card was involved.

graue70, 2023-04-13:
For me, images taken by LibreCamera appear in the gallery on the very bottom as if they didn't have a date or time. However, the Details of the image clearly show a timestamp.

Are you sure that the image doesn't appear in the gallery at all, @mvglasow ? Have you scrolled all the way down?
In that case, I will open a new issue.

mvglasow, 2023-04-13:
Indeed, the images appear at the very bottom, after the oldest camera images and together with other images that don’t have a timestamp. So we might want to examine the EXIF timestamp.