iakdis/librecamera #64: [Bug] No audio on videos taken

JuliettaExpress, 2023-03-08:
**Describe the bug**
Took a short video as a test, no sound was recorded. Not sure if relevant, but I had torch mode on for the video and shutter sound disabled.

**Other Information:**
 - App version 1.6.0
 - Device: LG V60 [LM-V600VM]
 - OS: Android 12

iakmds, 2023-03-08:
Hey there, thank you. Indeed, I noticed this a couple of days ago - it happens when you switch from photo to video mode, the camera isn't reloading. A patch release is on the way.

I will notify you here once it's on F-Droid

iakmds, 2023-03-21:
Version 1.6.1 is now on F-Droid, which should fix this bug. It took a bit longer because of an F-Droid related problem.

Let me know if I can close this issue.

closed 2023-05-11 by iakmds