iakdis/librecamera #65: [Feature request] Add leftet handed mode

dboehmer, 2023-03-22:
**Describe the solution you'd like**
When turning the phone to take a picture the camera button moves to the right. This is annoying when holding the phone with your left hand only.

I suggest adding a setting *left handed mode*. If enabled the camera button should move to the left when the phone is rotated horizontally. I couldn’t find such a setting in the app.

iakmds, 2023-03-22:
Thank you for the issue. 
Sounds like a great option, It'll be implemented in the next version

iakmds, 2023-04-24:
This issue should now be resolved with the release [v1.7.0](https://github.com/iakmds/librecamera/releases/tag/v1.7.0)! 
It should come out on F-Droid soon as well. Let me know if I can close the issue.

closed 2023-04-27 by dboehmer