iakdis/librecamera #68: [Feature request] Switch to Material You with dynamic colors

meenbeese, 2023-04-22:
**Describe the solution you'd like**
Even though Material You has been partially implemented as mentioned in #60, proper implementation is not here. What I mean by this is that the application does not change its colors and overall theme based on the color of the wallpaper. Blue is the default and only color available currently, which is not enough given that Material You is all about customization and personalization.

**Additional context**
There are other great apps which implement proper Material You and are written in Dart. The first that comes to my mind is [Obtainium](https://github.com/ImranR98/Obtainium), which you can check its source code to help with the implementation in your own app.

iakmds, 2023-05-11:
Hey @meenbeese,
thanks for the issue! Also thank you for linking the example. I took a look at it will implement it in the next version :+1:

meenbeese, 2023-05-12:
Thank you for addressing the issue so fast. I can't wait for the next release to test it!

iakmds, 2023-05-19:
Hey @meenbeese again,
This issue should now be resolved with the release [v1.9.0](https://github.com/iakmds/librecamera/releases/tag/v1.9.0)!
It's available on [F-Droid](https://f-droid.org/packages/com.iakmds.librecamera/) as well now. Let me know if it's working as expected.

meenbeese, 2023-05-19:
Hello @iakmds! I tried the new version 1.9.0 and it works perfectly. I like how the custom color is subtle and not aggressive. I also appreciate the AMOLED black theme as it looks great with the Material You colors. Thanks for implementing this.

iakmds, 2023-05-19:
I'm really happy to hear that! 
I assume we can close this issue now :smiley:

closed 2023-05-19 by iakmds