iakdis/librecamera #73: Switching cameras while videoing bugs footage

HarryHelsing, 2023-05-28:
If you start filming, then switch the camera, the footage becomes stuck after that point. Also if you pause then switch.
Perhaps this is a technical limitation and if so consider disabling the ability to switch camera's while filming.
Thank you!

Teloshav, 2023-09-06:
This here is the single reason I haven't been able to use this app yet. I've been looking around for some time for any open source apps with this feature and Librecamera is the closest I came across.

I'll give some information about what my situation's been like...

This is the tabulated notes I've made for when **recording horizontally**: 
| Resolution | Rear facing camera alone | Front facing/selfie camera alone | Rear-to-selfie camera flip | Selfie-to-front camera flip |
| :---: | :--- | :--- | :--- | :--- |
| MAX<br><br>(I think it is 3,840 x 2,160) | The footage is fine when looking back. The viewfinder displays the camera view smoothly.  | The footage comes out fine when looking back.<br><br>The viewfinder stutters and appears shaky before recording (same with the photos from the front-facing camera, but that's another issue) but weirdly enough is perfectly smooth once I start recording.  | There was no issues with the viewfinder while recording and pressing the button, no lag or stuttering of any kind.<br><br>The footage shows fine when the start of the video has the rear-facing camera just recording as normal, no issues there, but then when I go to swap the camera to the front-facing camera the footage just freezes at the last frame from the rear-facing camera.<br><br>There are no issues with the audio while all this is happening  | Just like the rear-to-selfie flip, the initial part of the video is fine in all regards, the viewfinder isn't lagging at all and there's no issues at this point when looking back on the footage either.<br><br>When I go to flip the camera to the rear-facing camera there is no issues on the viewfinder but there are weird issues when looking back at the footage.<br><br>The footage from the rear-facing camera _does_ show up but at a 90º rotation (as if I was holding the phone vertically) and was 'squeezed' so that the full view would still show up in the camera. The footage (_not_ the viewfinder) is also very stuttery/laggy(?) once I flip the camera. |
| LOW<br><br>(320 x 240) | No issues. Viewfinder and footage is as expected. | The viewfinder is shaky but stops when I start recording. | The viewfinder had no issues before and during recording.<br><br>When I flipped the camera the _exact_ same 90º rotation and 'squashing' happened, except there was no stuttering or lag. No audio issues either. | Pretty much identical to the rear-to-selfie camera flip. The viewfinder was stuttering a bit still (once again, this is likely another issue completely).<br><br>The footage had a 90º flip as well and is the exact same as all other occurrences, once again, this time had no lag or stuttering of any kind. |

Here's an example of what the 90º flip looks like (believe it or not, I didn't take this (or these?) photos! 😱): 
| BEFORE | AFTER |
| --- | --- |
| [image] | [image] |

I hope this data is useful to someone. As far as I've noticed, I think that it's related to both the orientation of the phone and the cameras and also the communication of the resolution/dimensions between the cameras when swapping. The lagging I assume could also just be some optimisation issue, but I think it's just a byproduct of the possible 'miscommunication' of resolution/dimensions.

Anyone else got a clue for why/how exactly this happens?

- Telsohav

Teloshav, 2023-09-06:
Just had a look at the [second part of the commit for the feature](https://github.com/iakmds/librecamera/commit/c32b589b2fa1b40e9a579eecdb4f33375e39e169) and I think the 90º rotation might be from this line here but this is just a complete guess from my part:
[image]

I just think to myself about 3 rotations being 270º or -90º as I noticed in my observations. Hope this helps!

- Teloshav