iakdis/librecamera #77: Unable to give Storage Permission

Appliumus, 2023-08-21:
**Describe the bug**
After installing i am unable to give the app storage permission. I tried to give it manually in the OS settings but there i also can not find an option to give the app permission. Only for Taking Photos/Recording/Mic 
I guess this bug is more a problem of the OS but i am not sure.

**Other Information:**
 - App version 1.9.1
 - Device: Pixel 6 Pro
 - OS: Android 13 / GraphenOS

TheSiman, 2023-08-22:
I have the same issue on Fairphone 4, CalyxOS (A13)

bnsmb, 2024-01-13:
Open  a shell (either on the phone or via adb) and issue

 pm  grant com.iakmds.librecamera android.permission.WRITE_EXTERNAL_STORAGE

That will fix the issue (the command can be done by the standard user -- no root access necessary)

saus23, 2024-01-23:
tried the command in termux and termoneplus, shows "Failure calling service package: Failed transaction (2147483646).
pixel 7a, Android 14, grapheneOS

ojde, 2024-04-01:
Same here, Samsung Galaxy Tab S8, Android 14
Cannot choose external sdcard, no way to give permission in settings, using `pm grant com.iakmds.librecamera android.permission.WRITE_EXTERNAL_STORAGE` in termux reports ` "Failure calling service package: Failed transaction (2147483646).`

van-de-bugger, 2024-09-03:
It seems I faced the same issue: I can't manage to save photos on SD card. 

In the first screen of the initial setup dialog pressing "Give storage permission" button does not have any effect.

In the second screen of the initial setup dialog, I press "Choose path" button, the file manager is open, I choose the folder ("Camera"), and press "Use this folder" button, in the dialog "Allow Libre Camera to access files in Camera?" I press "Allow" button.

Everything looks ok, but any attempt to take a picture ends with the error 

    PathAccessException: Can't copy file to '/storage/0000-0256/Camera/IMG_*.jpg', path = '/data/user/0/com.iakmds.librecamera/cache/*.jpg' (OS Error: Operation not permitted, errno = 1)

Attempt to use command `pm grant` does not help. I tried to enter the command in the Termux window, it fails:

    $ pm grant com.iakmds.librecamera android.permission.WRITE_EXTERNAL_STORAGE
    cmd: Failure calling service package: Failed transaction (2147483646)

Running the command through adb:

    $ adb shell pm grant com.iakmds.librecamera android.permission.WRITE_EXTERNAL_STORAGE; echo $?
    0

seems to complete successfully, but with no effect: Libre Camera still fails to copy images to the selected folder.

My phone is:
Zenfone Max Pro M1
Android version 13
LineageOS version 20-20240727-NIGHTLY-X00TD