iakdis/librecamera #103: [Bug] Shutter sound at high volume

c64whiz, 2024-06-23:
Greetings,

I just installed (via F-Droid) version 1.9.1 on a Samsung A13 5G running Android 13. 

While all my volume settings are down at about 10-15%, the shutter sound when taking a picture is still loud.  I _can_ disable the shutter sound and that works fine, but I'd prefer to have some audio or tactile (vibration) feedback a photo is taken.

I also tried turning off "Capture at volume press".  This 1) required a restart of the application to take effect (not really important), and 2) allowed me to verify media volume was low using the volume buttons.  However, taking a picture (via screen camera button) still had the shutter volume loud.

Any ideas?

sosasees, 2024-09-24:
even with every volume slider of Android set to silent, the camera shutter does not become silent.

i would like to set the shutter volume through Android's "system" category volume slider

iakdis, 2025-12-25:
Hi all, 

the app uses [MediaActionSound](https://developer.android.com/reference/android/media/MediaActionSound) to perform a `SHUTTER_CLICK` sound. There isn't much we can do about the volume unfortunately. However, version [v2.0.0](https://github.com/iakdis/librecamera/releases/tag/v2.0.0) now adds haptic vibrations!