iakdis/librecamera #107: Des not prompt for Storage Permissions cannot save anything

sojkaware, 2024-10-19:
**Describe the bug**
No photo or video can be saved due to app not having permissions. The button to give storage permissions in the welcome screen cannot ve clicked and the permissions cannot be added later. So unusable on my phone.

**Other Information:**
 - App version: 1.9.1
 - Device: Google Pixel 4XL
 - OS: Android 13

treyus30, 2024-12-05:
I'll add to this, the initial screen greyed itself after photo permissions, then after going into settings and both "select this folder" and re-initiating the beginning screen, still am unable to grant permissions. Graphene OS, if it matters.

iakdis, 2025-12-25:
Hi all, this should now be fixed in [v2.0.0](https://github.com/iakdis/librecamera/releases/tag/v2.0.0)!